mattdibi/redox-keyboard #97: only the side connected with usb cable works

schrodinger-s-cat, 2021-03-31:
Issue description: only the side connected with USB cable works, the other side does not work. Both sides works fine when connect with usb cable, the right side keys are mirrored of left though.

Debug done: I checked that the 4.7k ohm resistors are fine. The connection between PD0 of two pro micros checks out, as well as for PD1. After plugged in USB cable, both pro micros are powered and green LEDs light up. I am using the QMK toolbox to flash both side and used the default hex file from the toolbox.

As both sides work fine by themselves, and the connection between PD0 PD1 pins checked out, I am not sure why this does not work. Any ideas? Thank!

yterhorst, 2021-03-31:
Hi,

I have also just finished building my redox so this is just a guess.
The 4.7k pull-ups are only needed when using I2C which is not configured in the default QMK configuration (so probably neither in the precompiled hex file you are using). Instead the bit bang "serial" driver is used which should be using internal pull-up resistors. I don't see how the lower resistance would cause a problem, but if you want to make sure, you might want to either configure I2C in the firmware or desolder at least the resistor connected to pin 3 / PD0 of the Pro Micro (which is used for the serial connection).

schrodinger-s-cat, 2021-04-01:
I tried desolder the resistors and it did not help with the problem. Trying the I2C route at this moment. Actually had the same problem you have in your other post.

mattdibi, 2021-04-02:
I suggest you to take a look at the [QMK documentation](https://beta.docs.qmk.fm/using-qmk/hardware-features/feature_split_keyboard) and [this troubleshooting](https://github.com/nicinabox/lets-split-guide/blob/master/troubleshooting.md#troubleshooting) guide for the Let's Split keyboard (which uses the same machanism as the Redox for working).

Double check you're not using [the infamous Pro Micros described here](https://www.40percent.club/2017/09/green-pro-micro.html).

Copying from the Redox documentation:

>Pay attention to the Pro Micro you use in this project. Online you can find the infamous Green Pro Micros which have the same pinout of the normal Pro Micro but use a different power circuitry. Being the Redox a split keyboard that relies on detecting which side is directly powered to find which side is the master, these clones are not suited for the task. A more detailed explanation here. Please note that they can come in different colors, not necessarily green.

schrodinger-s-cat, 2021-04-03:
Confirmed I am not using the green pro micro. I desoldered the resistors and tried using the default hex file, still does not work. I am going to try the I2C route with 4 10k resistors.

mattdibi, 2021-04-04:
@schrodinger-s-cat Also please double check your TRRS cable. It was source of trouble in the past (#77 )

ahpook, 2022-12-27:
I had the same issue and the `#define SPLIT_USB_DETECT` resolved it - is this something that needs to be in the main docs? I am not sure what conditions cause it to be needed.

sidenkov, 2023-02-11:
encountered a similar issue. Built a rev1 without rgb, flashed the default layout, only the plugged-in half worked. `#define SPLIT_USB_DETECT` solved the issue