mattdibi/redox-keyboard #98: Half Without I2C Pull-up Resistors Is Unusable by Itself

yterhorst, 2021-03-31:
Hi,

I have built the wired version of the Redox and so far it's really fun to type on. I have found a small quirk however and was able to fix it. It doesn't seem to be referenced anywhere in the documentation. Maybe it would be helpful to add it :)

If I configure `I2C` and `EE_HANDS` (other than that I am using the default configuration) in the latest QMK firmware and connect the half without the pull-up resistors to USB without connecting it to the other half, it locks up and barely outputs anything or if it does, multiple key presses.

The culprit seems to be the floating I2C bus. Either spreading the resistance over both halves (using 10k resistors in all 4 footprints) or activating the internal pull-ups on PD0 and PD1 fixed the issue for me.

schrodinger-s-cat, 2021-04-01:
I actually have the same problem when configuring the keyboard with I2C and EE_hands, can you elaborate how did you activate the internal pull-ups on PD0 and PD1? Thanks!

mattdibi, 2021-04-02:
> If I configure I2C and EE_HANDS (other than that I am using the default configuration) in the latest QMK firmware and connect the half without the pull-up resistors to USB without connecting it to [...]

I didn't encountered this behaviour before. Maybe they changed something in the latest version of QMK and internal pull-ups are not activated by default?

I'll try and take a look into it. Thank you for pointing this out.