mattdibi/redox-keyboard #104: Hotswap Compatibility with Wireless PCB

TrapTrancer, 2021-04-11:
Hello, I have 3 questions about the wireless PCBs:
1) Does the kailh hotswap socket on the version 2 pcb have to specifically be the one mentioned in the BoM?
2) Can I use other hotswap options like copper rivets?
3) will https://www.aliexpress.com/item/4000333694486.html work on the receiver?

Thanks for the help as I am pretty new to keyboards still.

mattdibi, 2021-04-11:
Hi,

1. PCB Rev2.0WHS accepts Kailh Hot-Swap sockets (CPG151101S11), other types of hot swap sockets (like K4 Gaterons) were not tested at the time of this writing.
2. Copper rivets (a.k.a. Holtite sockets, a.k.a. Mill-Max sockets) are not compatible with Rev2.0WHS. They might be compatible with Rev.1.0W but I haven't tested them nor I intend to do so (I designed Rev2.0WHS to this exact purpose). The Holtite sockets mod is documented [here](https://github.com/mattdibi/redox-keyboard/tree/master/redox#hot-swappable-switches) for the non-wireless Redox.
3. Nope.

Hope this helps.

TrapTrancer, 2021-04-11:
Thanks for the help, I think I'll try the rev1 pcb with the copper rivets then.
I know this next question is a lot to ask, but do you know of a cheap (preferably $10 or lower) microcontroller that won't have the USB port ripped off?

closed 2021-04-16 by TrapTrancer