mattdibi/redox-keyboard #105: Right hand not working.

SirRonBird, 2021-04-19:
Greetings. 
First of all i wanna say thank you for this awesome project. It was much fun to build this keyboard. 

until now everthing works fine but after i flashes the boards, with the default firmware, the rightside of the keyboard dont work. I use a TRRS-Cable. (triple checked). I am a little bit confused, cause I dont know wether i have to solder the resistors, even if I dont use the rgb lights. (I´ve soldered them on). Could this be the Issue?

mattdibi, 2021-04-20:
Hi,

first of all, do they work independently? (i.e. if you plug in the right side, does it work?) Did you burn the firmware also on the right side? Did you check continuity on the TRRS plugs to the Arduino? Did you check the other already opened issues? ( #97 #77 #71 )

SirRonBird, 2021-04-20:
Hi. Yes they work independently. If I plug either side to the Computer both work. Just the otherone dont. 
Yes i burned the firmware on both (both have the same firmware !?) 
Yes i checked the TRRS Plugs. I could messure a through put from the plug from the one side to the pin of the other side. 
Yes. In the other issues they had an other Problem.

mattdibi, 2021-04-21:
If they work independently there must be something wrong with the link between the two halves.

> Yes i burned the firmware on both (both have the same firmware !?)

Yes, they both have the same firmware. Each side scans the matrix but only the plugged side send the key codes.

> I could messure a through put from the plug from the one side to the pin of the other side.

If you soldered the 4.7k resistor you should use the I2C communication. Did you enable it? Otherwise you can remove the resistor and use the serial communication (the default configuration). Keep in mind that I2C uses two wires for data, did you check continuity on both?

SirRonBird, 2021-04-22:
What exactly you mean with two wires? I have the TRRS-Cable plugged in and tested all poles in it.

I already tried with i2c enabled...

mattdibi, 2021-04-22:
> What exactly you mean with two wires? I have the TRRS-Cable plugged in and tested all poles in it.

Every pole carries a wire. An I2C connection used 4 wires or connections: VCC, GND (aka power), SDA, SCL (aka data).
Did you check that SDA, SCL are correctly wired through the two halves by the TRRS connector?

Follow this thread for more informations: https://github.com/mattdibi/redox-keyboard/issues/33

Ensure that, when the TRRS cable is connected, the PD0 pin of the left Arduino is connected to the PD0 pin of the right Arduino. The same goes for the PD1 pin.

Also: does the right hand Arduino lights up when connected to the left hand?

SirRonBird, 2021-04-22:
Ok. I checked the cable again and it seems that the tips of both ends are not connected. I will buy a new cable and try it again
The other side lights up...

mattdibi, 2021-04-22:
> Do you have a tip were i can get a custom TRRS Cable?

There are a few vendors around. Here's a short and (surely) incomplete list.
- [Coolcables](https://www.coolcable.co.uk/gb/trrs-cables/61-10630-custom-audio-cable.html#/coil_option-no_coil/double_sleeving_options-no_double_sleeving) (UK)
- [ZapCables](https://zapcables.com/diy-trrs-cable-kit/) (USA)
- [Kriscables](https://kriscables.com/product/trrs-cables/) (EU)

I used Coolcables in the past but they're out of stock in the TRRS department right now unfortunately

SirRonBird, 2021-04-23:
I just got the new cable hooked up and it still not working... I´ve checked the cable and the continuity between PD0 Pins and between PD1 Still only one side is working.... I down know why...

SirRonBird, 2021-05-10:
Hello. Ive rechecked everything and its still not working. The new cable is fine and the Pins SDA and SCL are connected. Does anyone have an other idea what the problem can be ?

SirRonBird, 2021-05-20:
So I worked a littlebit on the keyboard and it turns out, that something was wrong with the pcb. I build the right side again and now everything works.

closed 2021-05-20 by SirRonBird