mattdibi/redox-keyboard #107: Transmitter and receiver are not communicating

YoavGaudin, 2021-04-24:
Hello Matt,

Thank you so much for this awesome project.

I have almost completed the assembly of a redox-wireless keyboard, and I wish to test it before I solder all the switches (since I have to remove them from my currently working keyboard).
I have successfully flashed receiver and transmitter modules, as well as the QMK. I then soldered several switches to the left side hoping it will indicate whether things are working or not. But it didn't work.

So now I have a few questions-
* should the keyboard work if only some of the switches are connected?
* should the LEDs that we solder on the receiver be flashing when the Arduino is connected to the PC? (they are not and I fear I may have made a mistake with the leds direction)

Thank you for your help

mattdibi, 2021-04-24:
Hi @YoavGaudin,

> should the keyboard work if only some of the switches are connected?

Yes it should.

> should the LEDs that we solder on the receiver be flashing when the Arduino is connected to the PC? (they are not and I fear I may have made a mistake with the leds direction)

They should if, having uploaded the default keymap, you press the layer switching keys. More on that here: https://github.com/mattdibi/redox-keyboard/issues/93#issuecomment-808697030

I suggest you to test the keyboard before soldering the switches. You can do so by grabbing a pair of pliers and shorting the switch pads with them. That simulate a keypress.

YoavGaudin, 2021-04-25:
Thank you @mattdibi 
Unfortunately, it doesn't work. I'll have to purchase some testing equipment to figure it out.

matthiasdold, 2021-08-15:
Hello  @mattdibi  and @YoavGaudin,

@YoavGaudin  - did you by any chance figure out how to get it running? It seems, I do have a similar issue with everything being flashed ok (both manually and via the docker containers seems to work smoothly), but I too cannot get any key press or LED to light up on pressing the combination for the layer switch  (#93).

@mattdibi - thanks a lot for his awesome project! 
Is there maybe a way to debug the individual components? Looking at the YJ-14015 transmitted and received data via the ST-Link clone (some serial console from the debugger)? 
I noticed that on the receiver code there is [some lines](https://github.com/mattdibi/redox-w-firmware/blob/master/redox-w-receiver-basic/main.c#L126) which would provide a bit of insight. Would you know, what I would need to do to read this print via the ST-LinkV2 into a terminal/console?

Many thanks and have a nice week!

YoavGaudin, 2021-08-17:
Hi @matthiasdold 
Unfortunately I haven't found the time to work on that.
The last thing I checked was voltage levels with the circuit schema in mind, and everything seems to be in order.

I have been meaning to flash the transmitters once again to make sure I haven't confused left and right but didn't get around to it.

@mattdibi I will also appreciate some tips on how to debug the transmitters and receivers if possible.

THanks

matthiasdold, 2021-09-13:
Hi @YoavGaudin,

I made tiny little steps towards debugging:
- soldered a cable to pin 5 of the YJ-14015 on the receiver
- included a printf in the inner most loop of the receivers firmware - [around here]( https://github.com/mattdibi/redox-w-firmware/blob/master/redox-w-receiver-basic/main.c#L126)
- used my plain arduino uno with the following code:
````c
String data = "";

void setup() {

  // Open Serial Port
 Serial.begin(1000000);
 delay(50);
 Serial.println("\nStarting Up the serial monitor -> not line ending here, else it would be kept through the loop");
}

void loop() {
  
  // If there is data to grab, read it byte by byte
  char character;
  
    while(Serial.available()>0)
    {
      // taken from: https://electronics.stackexchange.com/questi 2 ons/84602/arduino-what-is-the-best-way-to-receive-and-read-data-with-softwareserial

      char character;
      character = Serial.read();
      data.concat(character);

      if (character == '\n'){
            Serial.print("Received: ");
            Serial.println(data);
            data = "";
       }

    }
  delay(5);
}
````
- powering the receiver from the pro-micro

When the wire from the receiver is connected to the arduino uno's RX pin, I would get the print statements on the serial console at /dev/ttyACM0 (just using the arduino IDE).

With this I noticed that the main() on my YJ-14015 starts over and over again after doing 4 iterations of the inner loop.... Will have to investigate further once time allows. 

Also I think I bricked my YJ now by inserting the powered pro-micro shifted by one row resulting in RAW -> GND on the receiver which probably is not good, so be aware.

Aztriltus, 2021-10-09:
Hey @matthiasdold @YoavGaudin have you guys solved the issue of the receiver and transmitters not communicating? 

I have a similar issue where my LEDs do not light up and any key presses (including the default keycap layer keys) do nothing despite all the firmware being flashed without failure. 

Is there any way to debug the issue? I have made 3 receivers just to be sure it’s not my soldering issues and all of them were able to flash without failure + QMK is also fine since my computer is able to detect them via QMK toolkit as redox wireless. Tried both the compiled redox_w and loading the pre compiled. Shorting my transmitters and doing a continuity check works so the battery + kailh sockets are fine + I was able to flash both the transmitter. 

I was a bit confused about the LEDs for the receivers but I’m wondering if this will affect the communication between the receiver and transmitters?

Any ideas on what I could do? @mattdibi

YoavGaudin, 2021-10-10:
Hi @Aztriltus 
Unfortunately I haven't been able to solve this.

All I tried was to flash of the components once again making sure I am not confusing left and right. But it is still not working.

Furthermore, I don't have the proper equipment to really get to the bottom of this (desoldering, monitoring etc...)

In the mean time, I have ordered a pre-built split keyboard (not wireless) and I'm trying to sell the parts of the wireless redox.

matthiasdold, 2021-10-10:
Hi @Aztriltus - unfortunately my day job leaves me not time atm to work on this...

I did take a raw nrf51 without the pcb and the LEDs etc, soldered a few wires to it and just hooked this up to my arduino printing out the matrix with the code mentioned above. At least from what I could see there, the matrix looked stale, thus my next steps would be to hijack a transmitter nrf51 see if and what it would send.

Definitely not giving up on this, but it will take a few more months for me :(

shenyingzxc, 2022-01-15:
Hi @matthiasdold just wondering if you’ve had any progress on this issue since your last reply? I have the same issue as you and no arduino, so I haven’t been able to try your troubleshooting steps… Running out of ideas on how to make this work, haha.

matthiasdold, 2022-01-16:
Hi all,
unfortunately no breakthrough yet, but a big step towards it. I figured out, how to run GDB on the yj-14015 and am now sure, that my last problematic communication is between the receiver yj-14015 and the arduino.

The most helpful video I did find along the way is [this](https://www.youtube.com/watch?v=OYxEWDpw0g8) - in German unfortunately, but he is so slow an deliberate, that one should be able to follow.

Finally my debugging setup looked like this:
[openocd_setup.pdf](https://github.com/mattdibi/redox-keyboard/files/7877131/openocd_setup.pdf)

Along the way I met several issues, e.g. the Arduino setup above helped to identify that I still have a few diodes soldered the wrong way. Also notice that there is no pin mapping between the yj-14015 and the nrf51822 with the build flags used in the redox firmware repo. -> I spend a few hours getting a blinky to run thinking I would need a pin mapping :(

Hope this helps some of you. Once I figure out the last bit, I will try to do a more comprehensive write down of my trouble shooting steps.

Best regards,
Matthias

ErichKinder, 2022-01-28:
I'm also having this issue. I've tried multiple receivers with multiple pro micros and none of them appear to be working.

buddylindsey, 2022-02-22:
I think I am hitting this issue.

Everything said it flashed properly, but when I plug in the reciever nothing happens. I get a green led that comes on. So I assume it is starting, but after that nothing when I hit keys.

I double checked all my diodes and they are good. Just no idea how to tell if anything is working.

mattdibi, 2022-02-23:
@buddylindsey please open a new issue describing what you did and what's not working. I'll follow you on that.

buddylindsey, 2022-02-23:
So one final update. I am now typing this on my redox V2WHS keyboard. I did a bunch of testing with a meter and things seemed to check out. 

I ended up taking apart all of the pins on the receiver to arduino pro and resoldered them to fix the connections. Once I did that everything started working properly. Thank you @mattdibi for all the help and for making this project.