mattdibi/redox-keyboard #109: Wired connection between wireless halves and reciever

eppingere, 2021-05-03:
Hey,

I have been having connection issues with my wireless redox. I am in the process of debugging them, but if I cannot get them resolved is there a way to connect both halves to the receiver with a wire. Effectively making it a regular wired redox? If so what wire should I use and do I need to change firmware at all or is this already supported?

mattdibi, 2021-05-03:
Hi,

> is there a way to connect both halves to the receiver with a wire. Effectively making it a regular wired redox? If so what wire should I use and do I need to change firmware at all or is this already supported?

Probably yes there is a way, but you are essentially re-engineering everything from the ground up. You would need to rewrite the firmware from zero or port QMK over to the YJ-14015 (ARM Cortex-M0). Not something I would suggest.

eppingere, 2021-05-04:
Sad :(

Thanks for the response

closed 2021-05-04 by eppingere