mattdibi/redox-keyboard #111: Keyboard stopped working after a short while

Andrea-Perin, 2021-06-26:
Hi
I have built two Redox keyboards, but one of them is having major issues.

Issue description: after I flashed it with QMK toolbox's auto flash, the keyboard worked properly.
However, after some 5 minutes or so, it stopped working. The LEDs were on, but the keyboard wasn't being detected.
After some hours I tried it again, and this time it worked. However, the next day, it alternated between working and not being detected, in bursts of some minutes.
I can not re-flash it, as it is not being detected by the computer, so I am stuck with no apparent reason for this behavior.

Preliminary checks: I checked all the soldering points, and there are no obvious issues there. Both Pro Micros should be fine (or at least, they were at the time of the first flashing). The cables I am using are also working with the other Redox.

Do you have any ideas about what could be causing this?

Btw, thank you for the project!

mattdibi, 2021-06-27:
Ciao Andrea,

this is a new one.

Did you try plugging in the USB in the other half? If so, the behaviour is the same for both halves? Did you check the USB cable? Have you tried pluggin the two halves independently (i.e. without the TRRS cable)?

> The LEDs were on, but the keyboard wasn't being detected.

Are we talking about Pro Micro's LEDs or backlighing LEDs?

Andrea-Perin, 2021-06-27:
Ciao,
thank you for the prompt reply. To answer your questions:

- When it does not work, plugging either half makes no difference, it is simply not detected (if it can be of any help, I did not set any value for the master hand flag in the header file, so if I am not mistaken, the left should be the master).
- The USB cables I am using are both working with no issues with the other Redox I built; they are USB cables that came with a phone, so I would assume they are fine.
- As for plugging them separately, I did not try that.
- I was referring to the Pro Micro's LEDs, there are no backlighting LEDs.

As a matter of fact, I wrote half this reply with the Redox itself, but I am ending it with the laptop's keyboard, as the Redox stopped being detected halfway through. 

I ran `lsusb` while it was working, and `Falbatech The Redox Keyboard` appeared among the outputs. Right now, instead, it's either not detected at all, or `Arduino SA Leonardo Bootloader' comes up.

mattdibi, 2021-06-27:
> it's either not detected at all, or Arduino SA Leonardo Bootloader comes up.

Interesting. Looks almost like the Arduino puts itself into programming mode by its own. Could there be a short in the RST (reset) pin area? Could you be sending the reset code via software by any chance?

> As for plugging them separately, I did not try that

Give that a try to rule out the cable or one of the Arduinos. By plugging in the slave half by itself it should output the mirrored matrix IIRC.

Also at this point it seems to be reproducible: after being plugged in it works for some time then it’s puts itself into programming mode. I would try to reflash the firmware just in case, but it seems to me that there might be a problem with the master half Pro Micro.

Andrea-Perin, 2021-06-27:
Ok, so I tried the two halves separately: it seems that the right one works fine (as you said, it mirrors what is supposed to be one the master side), and the left one is the problematic one.
I tried removing the reset button on the left half, hoping it was defective and thus causing all the problems, but no luck there. Still the exact same problems.
As for sending a reset code via software, there is no reset key in my mapping, so unless there is something doing it without me knowing, I would rule that out.

I was wondering, could it be due to bad soldering? Like, maybe some overheating due to imperfections in the way it was soldered. I say this because once it stops working, there is no guarantee that it works if you plug it back right away. So maybe there is something more "physical" about it?

Either way, I will try reflashing it when I have time, and see if that helps. Thank you so much for now, I will get back to you with the updates.

mattdibi, 2021-06-27:
> So maybe there is something more "physical" about it?

Yeah, by the looks of it I would say that this is a hardware issue with the left hand Arduino.

Try giving another pass to the Arduino soldering joint, maybe adding some more solder, just to be sure.

Andrea-Perin, 2021-08-01:
Hi again, I am back with some good news! I finally got my hands on a working soldering iron yesterday, so I tried re-soldering the left hand side Arduino. Interestingly, most of the soldering points "fizzled", as if some air bubbles were trapped inside them. When I plugged the keyboard back in, it worked. I have been using it for the past 2 days and I have yet to notice any problems.

Most likely, it was a matter of bad soldering joints, as you suggested. I will close the issue, since now everything seems fine.
Thank you so much!

closed 2021-08-01 by Andrea-Perin