mattdibi/redox-keyboard #112: Odd rollover issue

JamesYeoman, 2021-06-30:
After going through hell attempting the handwire version, I finally decided to buy some PCBs. I've finally gotten to a firmware flash for the left hand, and everything seems to be working great. However, there's some weird rollover that I've noticed.

When holding D down and pressing A, it rolls over correctly. So far so good.
When holding F and pressing S, it rolls over to S, and then rolls over to D for some strange reason. And it's the entire column that it happens on. The rollover has to occur on the same row (it doesn't freak out when R is held and S gets pressed). Holding 4 and pressing 2 results in a 2->3 rollover.
V+X => X->C
B+C => C->V
And so on.
[+F doesn't result in F->G, however.

It's the columnar nature that makes me _fairly confident_ that it's not a soldering issue. It wouldn't be that consistent of results if it were simply a soldering issue. And if it was the column pins being shorted, then it wouldn't matter whether the beginning key was on the same row as the rollover key.

As a sanity check, I did a quick sweep of the pcb and couldn't find anything out of the ordinary...

It's a weird one, this is...

mattdibi, 2021-07-01:
> It's the columnar nature that makes me fairly confident that it's not a soldering issue. 

Actually, the columnar nature of the issue is a clear indication that might be a soldering issue.
I'll borrow the troubleshooting guide from the Corne build guide:

> A full row/column of keys not working
If a full row or column of keys is not working, then the culprit is most likely the connection between the PCB and the ProMicro. Check your soldering and make sure there's a proper connection and that you have soldered the ProMicro in the right set of holes. If soldering looks okay, then your ProMicro might be damaged. You can exclude the possibility of problems with the PCB, paths and diodes by short circuiting the pins on the ProMicro directly, using a bit of wire. Connecting one row pin with one column pin should result in the corresponding key. Some PCBs have silkscreen print indicating which pin is which row or column, to make this process easier.

JamesYeoman, 2021-07-01:
Ah. Thanks. I'll have a look after work and give you an update

JamesYeoman, 2021-07-05:
I've had some issues with the desoldering, which has meant the destruction of the pro micro. It was the only one I had, too. I'm going to have to close this issue for now because I can't afford any pro micros until payday.

closed 2021-07-05 by JamesYeoman