mattdibi/redox-keyboard #113: "Cannot start service openocd" with macOS

daseintheworld, 2021-07-02:
Hi, I have tried programming both side of PCBs on macOS.
It fails at running containers with "docker-compose up" command.
I think error messages are related to udev.

Starting redox-w-firmware_openocd_1 ... error
ERROR: for redox-w-firmware_openocd_1  Cannot start service openocd: error gathering device information while adding custom device "/dev/bus": no such file or directory

 Do you think moving to a linux device is the only remedy? :(

mattdibi, 2021-07-02:
Hi,

yeah unfortunately it seems to be quite hard to do. There's [a weird VirtualBox trick](https://christopherjmcclellan.wordpress.com/2019/04/21/using-usb-with-docker-for-mac/) but I never tried it. This is the first link a found and can't recall what the actual query was so I suggest you to keep on digging.

Hope this helps.

tomasz-bochenek-perform, 2021-09-04:
In my case I've decided to roll back to the non-docker-based version in the repository: https://github.com/mattdibi/redox-keyboard/tree/1c5757389c0fc705b7f8c10168a5efc069b68620/redox-w and do everything "by hand". One note along the way: the mac os `readlink` command won't take -f params, so I just replaced it entirely with a hardcoded full path to