mattdibi/redox-keyboard #114: Requesting Kicad files for rev 2.0WHS

Bombergames, 2021-08-28:
It looks like the rev 1 kicad files were updated with the 2.0 commit, but it doesn't have the actual updated files for the rev 2.0 board.

jesper1266, 2021-09-20:
Hi, I second that :-)
I would like to help.
It would for example be nice tha the pinheader for uploading the firmware is the same for both the board and reciever.

Cheers
Jesper

mattdibi, 2021-09-26:
Hi guys,

unfortunately I'm having issues with recovering the files: my Windows installation decided to brick itself after an update and cannot access it anymore. I'll try and upload the files when/if I recover them.

PhiBabin, 2021-11-21:
Hi guys,
I rebuilded the KiCad of rev 2 by using the rev 1 as a starting point and using the gerber files of the rev 2 as reference. I included some of the modified footprints its own library.

KiCad project file: [redox_rev2_recovery_kicad.zip](https://github.com/mattdibi/redox-keyboard/files/7575625/redox_rev2_recovery_kicad.zip)
Gerber file from rebuild: [redox_rev2_recovery.zip](https://github.com/mattdibi/redox-keyboard/files/7575623/redox_rev2_recovery.zip)

Original gerber top layer:

[image]

Rebuild gerber top layer:
[image]

Original gerber bot layer:
[image]

Rebuild gerber bot layer:
[image]

This is my first time using KiCad, so use this at your risk and peril.

KeyTamer, 2021-12-17:
First thanks, a lot for the KiCad file! 
A other think. Did you test it all ready?

PhiBabin, 2021-12-21:
Yeah, I send it to AllPCB and they worked fine. I'm typing on it right now.
The only issues I had are : 
 - The diode direction is not visible because the pads are too close to the marking. However, from looking at the image of the original, it seems that it has the same issue. This doesn't really matter, since the guide said that you just have to oriented the diode toward the square pad. It should be super to fix that in a future version.
 - The holes for the hotswap are not circular, they are skew (see last picture). This mean that when you insert each hotswap you get a very satisfying *click*. It also mean that your fingers are going to hurt after putting 70 hotswaps. I don't think the kidcad project is the cause of this, this is probably an issue with how I exported the Gerber or with AllPCB or a normal issue when doing overlapping hole like this. Anyway, it really doesn't matter and it's a very minor issue.

Note : The version that I printed is different from the KiCad project files shared above, I added the option of putting a rotary encoder at the location used by key K16. I want to finish the QMK+firmware changes before sharing it.
[image]
[image]
[image]

widget-, 2022-04-24:
@PhiBabin I'm interested in trying this out and probably opening a pull request into this repo assuming it goes well. Can you confirm that you're releasing it under the same [MIT license](https://github.com/mattdibi/redox-keyboard/blob/master/LICENSE) as the original project?

PhiBabin, 2022-04-26:
Yes, I'm  releasing it under MIT license.
Sorry forgot to upload them a fork.

l4u, 2022-07-02:
@PhiBabin Do you have a forked version or a pull request please? I'd like to make modifications based on your version with proper git history.

PhiBabin, 2022-07-24:
@l4u  sorry for the delays... https://github.com/mattdibi/redox-keyboard/pull/151