mattdibi/redox-keyboard #115: Bootloader/reset mode not accessible anymore

bilange, 2021-09-13:
I am now apparently unable to boot into bootloader/reset mode. After an unsuccessful flash (more on this later), pressing the reset button on the PCB will make Windows say it's an unrecognized device. The keyboard will switch immediatly (without waiting ~10s in bootloader mode) into normal mode. 

However on Linux (Fedora) I noticed that the device will outright disappear from the lsusb device listing, and stay absent, once I press reset. A power cycle is needed to return to normal mode. 

Normal mode:
```
❯ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04ca:2006 Lite-On Technology Corp. Broadcom BCM43142A0 Bluetooth Device
Bus 001 Device 002: ID 04f2:b483 Chicony Electronics Co., Ltd USB2.0 VGA UVC WebCam
Bus 001 Device 009: ID feed:0000 Falbatech The Redox Keyboard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Once I pressed reset (either physically or via QMK RESET): 
```
❯ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04ca:2006 Lite-On Technology Corp. Broadcom BCM43142A0 Bluetooth Device
Bus 001 Device 002: ID 04f2:b483 Chicony Electronics Co., Ltd USB2.0 VGA UVC WebCam
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

## Unsuccessful flash (root cause)

The situation avove may be my own fault, as this happened right after a flash that apparently stopped mid flashing, and for both sides individually, at 88% for many minutes. Ctrl-C didnt abort, I had to physically unplug the keyboard to halt the procedure (I DO know about not doing it in normal circumstances, but as it appeared obvious flashing was somehow halted mid procedure, what else could I do?).

Console output (from Windows QMK MSYS): 

```
[bilange@warbird qmk_firmware]$ avrdude -p atmega32u4 -P COM3 -c avr109 -U flash:w:stable\ -\ backup_flash.hex

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude.exe: Device signature = 0x1e9587 (probably m32u4)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file "stable - backup_flash.hex"
avrdude.exe: input file stable - backup_flash.hex auto detected as Intel Hex
avrdude.exe: writing flash (32730 bytes):

Writing | #####                                              | 10% 0.28s
(...)
(skipping many lines)
(...)
Writing | ############################################       | 88% 2.45s
(stuck indefinitely there)
```

While flashing seems to work at first sight (the "stable - backup_flash.hex" file was succesfully applied as my keyboard seems to act as per this file*), it has the side effect of having somehow wiped (partially??) bootloader.

* : With the caveat that this file was the result of backuping a known good config file from the keyboard, per `avrdude -P COM5 -c avr109 -p atmega32u4 -U flash:r:backup_flash.hex:i`. I *do* usually `qmk flash` but have lost my stable .hex binary inbetween workstation reinstallation.

So, is there a way to get out of this situation? Mind you, I CAN use the keyboard, only NOT flash it anymore- so i'm basically stuck with whatever is on my keyboards memory right now.

Thanks!

mattdibi, 2021-09-25:
Hi,

I'm sorry but I can't help you here. I would redirect you over the QMK github repo for this kind of issues.

Cheers,
Matt

closed 2021-09-25 by mattdibi

bilange, 2021-09-25:
Hi Matt, 

It definitely looks like I need to reflash my bootloader according to other people having the same issue with other keyboards. I'm in the process of acquiring the proper wiring tools and an additional Pro Micro and will definitely try to flash it back. 

I understand this is somewhat unrelated for your Github project, but for the sake of your issues section completeness (for future redox readers that could stumble on the same issue), I will report back when my wiring+Pro Micro order is shipped, if you don't mind.

Sources (partial): 

- https://beta.docs.qmk.fm/using-qmk/guides/keyboard-building/isp_flashing_guide
- https://old.reddit.com/r/olkb/comments/prb7wp/keyboard_wont_flash_after_bad_hex_flash/

Thanks!

(Sidenote: I hope submitting my comment won't reopen automatically this issue.)