mattdibi/redox-keyboard #118: [Question] Possible to use bluetooth module with rev1 PCB?

mauedu, 2021-10-01:
I have a few PCBs for the redox rev1 and I would like to add a bluetooth module to one of them. I would like to connect a Adafruit Feather 32u4 Bluefruit LE to the PCB which has [hardware SPI capabilities](https://learn.adafruit.com/adafruit-feather-32u4-bluefruit-le/configuration).

From the [breakout diagram of the Pro Micro](https://cdn.sparkfun.com/assets/f/d/8/0/d/ProMicro16MHzv2.pdf), it looks like the SPI pins are on B1, B2, and B3. However, after looking at the QMK `config.h` file for the rev1, it looks like those [pins are being used to define the keyboard matrix](https://github.com/qmk/qmk_firmware/blob/master/keyboards/redox/rev1/config.h#L35-L36).

Is there a way to connect a bluetooth module to the rev1 PCB?

mattdibi, 2021-10-03:
Hi,

If you really want a bluetooth version of the Redox I would suggest to use [ZMK](https://zmk.dev/) and a pair of [nice!nano](https://nicekeyboards.com/nice-nano) or [bluemicro](https://nrf52.jpconstantineau.com/docs/bluemicro840_v1/) which should be compatible with the Redox rev1.

The pins you're looking are wired in such a way that doesn't allow for them to be used in other ways (i.e. they can only be used for matrix scanning and not SPI).

mauedu, 2021-10-03:
Understood. Thank you for the clarification.

closed 2021-10-03 by mauedu