mattdibi/redox-keyboard #122: - and + keys not working on one computer, but working on another.

BartStikkers, 2021-12-04:
Issue as described above. The +and - keys on the left hand bottom row don't work at all, the LALT_T(*) registers as a right mouse button. The top right _- doesn't work either. If I plug the keyboard into my laptop they all work correctly. I used the first PC to flash the keyboard. Doesn't matter if I use a USB hub or not, doesn't matter if I use USB 2 or 3.

BartStikkers, 2021-12-06:
That'd be really funny, actually. I haven't even used a keyboard with a numpad for years. I had a look though, there's nothing like that I can find. Maybe it has to do with the drivers QMK installed?

mattdibi, 2021-12-06:
As per [QMK documentation](https://beta.docs.qmk.fm/using-qmk/guides/driver_installation_zadig):

> QMK presents itself to the host as a regular HID keyboard device, and as such requires no special drivers. However, in order to flash your keyboard on Windows, the bootloader device that appears when you reset the board often does.

So those drivers shouldn't affect keyboard behaviour.

BartStikkers, 2021-12-06:
The only other differences are that the desktop is much older (i5-4590) and I connect it through a hub, but plugging in directly didn't make a difference.

I made a mistake in the report btw: it's the left hand +,- and * key. Holding down * does trigger ALT, and it's + that triggers a right mouse click. The top right key works, but holding down MO1 and hitting +, * and - then (under P, H and ;) yields the same result, so it really is like the computer isn't understanding the key codes right.

BartStikkers, 2021-12-07:
After deinstalling qmk and rebooting, the problem is now gone. Still don't know why this happened, but that's a solution at least.

closed 2021-12-07 by BartStikkers