mattdibi/redox-keyboard #123: keyboard stops working after some time.

BartStikkers, 2021-12-04:
See title. Unplugging the USB and plugging in again fixes it, but it happens often enough that the keyboard isn't usable. When it happens seems to have no connection to how long it's been plugged in or when it's last been used. It sometimes happens in the middle of typing.

mattdibi, 2021-12-06:
What type of Redox are you using? Wired or wireless?

BartStikkers, 2021-12-06:
Wired, the rev1 with the PCBs from Falba.

mattdibi, 2021-12-06:
Mh... this is really unexpected.

My first suspect would be hardware:
- Double check all solder joints for any kind of short, especially on MCU pins
- Could be a bad Pro Micro. Try using one half at a time and check whether the issue appears on only one half. This could also pinpoint where the (eventual) short could be

BartStikkers, 2021-12-06:
That was my first thought as well, but I can't find any shorts. I wouldn't be surprised if there was something wrong with these Arduinos though, they were really cheap. I've reflashed them just to rule out any issues from the first time and the problem hasn't reoccured. The only thing I think I'd changed though is the communication between the halves from serial to i2c, maybe I did something wrong there that caused it? I'll try using it tomorrow, see if the problem reoccurs.

BartStikkers, 2021-12-07:
Problem appears solved, so I guess I screwed something up when I was messing with the config files. Thanks for thinking along!

closed 2021-12-07 by BartStikkers

imalexlab, 2022-01-11:
I'm having the same issue, also i2c, do you remember what did you change in the files @BartStikkers ?

BartStikkers, 2022-01-15:
iirc:  qmk_firmware/keyboards/redox/keymaps/default/config.h I swapped the lines for USE_SERIAL and USE_I2C, commenting one and uncommenting the other.