mattdibi/redox-keyboard #125: Sourcing Parts

buddylindsey, 2022-01-11:
There is still a lot I do not understand on what particular parts I need. I am just taking what is listed in the Wireless parts and buying them.

I am left with the resistors, LEDs, headers, and momentary switch. Being surface mount and not understanding what some of the stuff is. I put together a chart where I am trying to source parts from to make sure if I order them I am ordering the correct.

Can someone double check me please? I don't want to order everything and waste time and money because I am being stupid and order something wrong.

Here is my table of parts and links:

|item|link|
|-|-|
|330 Ohm 0805(2012 metric) SMD resistors|https://www.digikey.com/en/products/detail/ERJ-6ENF3300V/P330CCT-ND/1746851?itemSeq=386817440|
|220 Ohm 0805(2012 metric) SMD resistors|https://www.digikey.com/product-detail/en/RCS0805220RJNEA/541-2862-1-ND/5867155?itemSeq=386817758|
|1.5k Ohm 0805(2012 metric) SMD resistors|https://www.digikey.com/product-detail/en/ESR10EZPF1501/RHM1.50KAECT-ND/1983817?itemSeq=386818029|
|Red 0805(2012 metric) SMD standard LED|https://www.digikey.com/en/products/detail/lumex-opto-components-inc/SML-LX0805IC-TR/515787|
|Blue 0805(2012 metric) SMD standard LED|https://www.digikey.com/en/products/detail/dialight/5990191007F/9385414|
|Green 0805(2012 metric) SMD standard LED|https://www.digikey.com/en/products/detail/sunled/XZMG54W-1/4745961|
|White 0805(2012 metric) SMD standard LED|https://www.digikey.com/en/products/detail/qt-brightek-qtb/QBLP631-IW/4814683|
|Through hole momentary switch|https://www.mouser.com/ProductDetail/CK/PTS645SK43JSMTR92-LFS?qs=TiOZkKH1s2TgjwNoJW7SNA%3D%3D|
|Right angle 0.1" header (1x4pin)|https://www.pololu.com/product/2704|
|Straight 0.1" header (1x6pin)|https://www.pololu.com/product/1016|

mattdibi, 2022-01-11:
Hi @buddylindsey,

from what I see everything looks in order.

A couple of notes:
- All the materials for the Redox Wireless are meant to be easily found on ebay for (usually) a cheaper price than Digikey/Mouse etc. Did you try and take a look there? Like you can get an assortment of leds (like I did) [here](https://www.ebay.com/itm/192211418194?hash=item2cc0b17c52:g:SnIAAOSwLsBZOUYA). The same goes for the resistors.
- I noticed that all your headers pin are female. Is this correct? Do you have the necessary jumpers for plugging everything into the ST-Link programmer?
- The LEDs (and related resistors) are not strictly required for the keyboard to work, you only _need_ R1, R2 and R3 resistors.

buddylindsey, 2022-01-12:
> Hi @buddylindsey,
>
> from what I see everything looks in order.
>
> A couple of notes:
>
> * All the materials for the Redox Wireless are meant to be easily found on ebay for (usually) a cheaper price than [...]

I did not realize that. I will check ebay out for sure.

> * I noticed that all your headers pin are female. Is this correct? Do you have the necessary jumpers for plugging everything into the ST-Link programmer?

I don't have an ST-Link programmer yet. I hadn't gotten that far in the purchasing process. I kind of am just copy/pasting and searching for what you have listed trying to find parts. So TBH not sure what all I need in the end.

> * The LEDs (and related resistors) are not strictly required for the keyboard to work, you only _need_ R1, R2 and R3 resistors.

I like the idea of the LEDs so I wanted to use them.

mattdibi, 2022-01-14:
> I think I have the male ends of the headers already. Is this what I need? If so I have them.

They're the one I was referring to.

> I don't have an ST-Link programmer yet. I hadn't gotten that far in the purchasing process. I kind of am just copy/pasting and searching for what you have listed trying to find parts. So TBH not sure what all I need in the end.

The ST-Link programmer is needed to upload the firmware to the YJ-14015 MCUs. See [firmware guide](https://github.com/mattdibi/redox-keyboard/tree/master/redox-w#firmware) for details.

buddylindsey, 2022-01-14:
Awesome thank you. I have a programmer on order and should be here today, hopefully. Now to wait on the rest of the stuff. Interestingly the pub boards from pcbway has proven to be one of the fastest things to come in.

Thank you for this it is going to be awesome.

buddylindsey, 2022-02-19:
Okay I am going to close this as I am pretty sure all of the parts have come in now. Hopefully I can start on this thing soon, we as soon as I get a new tool to arrive, heh.

Will also post up another question later I have on cases, and which cases to print.

closed 2022-02-19 by buddylindsey