mattdibi/redox-keyboard #126: ZMK firmware + nice!nano Redox build guide

mattdibi, 2022-01-16:
As of https://github.com/zmkfirmware/zmk/pull/1002 ZMK supports the Redox keyboard. Add a build guide using nice!nano and the ZMK firmware.

chrismelba, 2022-01-18:
I've just built one of these. What are you looking for exactly? Also, does ZMK support redox-w? Or just redox with a nice!nano?

mattdibi, 2022-01-19:
Hi @chrismelba

thank you for reaching out!

> What are you looking for exactly?

I would like to add:
- the bill of materials
- a small build guide with a few pics of the mounted nice!nanos
- a quickstart on how to upload the default keymap

for the ZMK-based Redox. Not much detail regarding the hardware (diodes, switches etc) there's already the [main guide](https://github.com/mattdibi/redox-keyboard/tree/master/redox#assembly-guide) for that.

> Also, does ZMK support redox-w? Or just redox with a nice!nano?

I don't think so. Only the redox rev1 supports the Pro Micro footprint. The redox-w uses the YJ-14015 (NRF51822) modules that I don't find in the ZMK docs among the supported hardware.

What do you think?

snizovtsev, 2022-02-06:
I've spent last New Year's holidays unsuccessfully trying to port ZMK to YJ-14015. The main problem here is that it comes with NRF51822 QFAA chip with 16KB SRAM. Its not enought to get Zephyr's Bluetooth stack work. Even with super-crazy hacks I've only managed to run a simple Zephyr example that can advertise itself with BLE. However it crashes on pairing with stack overflow.

But here is a good news:
- Zephyr's drivers are able bring NRF51822 BLE physical layer to work;
- There are NRF51822 QFAC revision exists with 32KB SRAM onboard!

I think I could get ZMK work with QFAC version of chip on my next approach.

snizovtsev, 2022-02-06:
As a hardware note, I'm not able to find any YJ-14015 modules with QFAC onboard. The only ones with 32KB of memory are original Core51822 which costs a slightly more. I've bought a few ones from [Waveshare](https://www.waveshare.com/core51822-b.htm) ([Aliexpress](https://aliexpress.ru/item/32374207080.html?sku_id=57373132936)) and another problem arise: they are a little bit wider than YJ-14015. However I've able to get it work with original PCB with help of hot air gun:

[image] [image]

Zolntac, 2022-03-03:
Guidance on making a redox1.0 that's wireless via nice!nanos and ZMK would be so useful to me, and assuredly others. Having the extra build information, firmware installation, and if you wanted to be kind, info on considerations battery wise, like voltage, connection, and other info.
Users also maybe get both wired and wireless to swap to and from via this means which would also be amazing. Though currently I dont know if that's possible when you make a Redox1.0 and then make it wireless via nice!nanos because of lack of info. There's just missing info on understanding and building this version of the keyboard. I would greatly value if it was added, as I intend to make it and I assume others will too.

bmcustodio, 2022-03-04:
I second this, it would be very useful to me too!

bmcustodio, 2022-03-06:
@mattdibi @chrismelba would the bill of materials and the guide differ from the main ones (except, of course, for the nice!nanos)? My understanding is that the nice!nano is a drop-in replacement for the Pro Micro, but I am fairly new to this. What else would need to change?

mattdibi, 2022-03-06:
@bmcustodio As you said the BOM wouldn't differ much: 

| Qty | Item                                          | Notes                                               |
|----:|-----------------------------------------------|-----------------------------------------------------|
|  70 | Cherry MX compatible switches                 |  Still required                                                   |
|   2 | Redox PCBs                                    | Still required |
|  70 | 1N4148 diodes                                 | Still required                        |
|   2 | PJ-320A 4 poles 3.5 mm TRRS connectors        |  **Not** needed                                                   |
|   2 | 4.7 kOhm resistors                            | **Not** needed                |
|   2 | Through hole momentary switch                 | Still required (albeit not strictly necessary)                       |
|   2 | Nice!Nano | Needed |
|   1 | TRRS cable                                    | **Not** needed                                                    |
|   1 | USB-C cable                               |   For charging and programming                                                  |
|  70 | Cherry MX compatible keycaps                  | 10x 1.25u keycaps, 6x 1.5u keycaps, 54x 1u keycaps  |
|  14 | WS2812/WS2812B leds                           | Would not recommend for a wireless build                             |

In addition to the above you need **two**

> 3.7v rechargeable lithium battery that is at least 100mAh large (you can probably get away with 80mAh). Note when choosing a battery, the recharge rate is 100mA, so a 2,000mAh battery will take 20 hours to charge. The overall recommended battery for most low power (no LEDs or extra power sinks) is the 301230 battery. This battery is 3mm thick and fits underneath a socketed nice!nano very nicely. These can be found on some vendors' stores as well as on AliExpress.

As per the [nice!nano documentation](https://nicekeyboards.com/docs/nice-nano/#recommended-batteries-and-sockets).

In addition I highly suggest socketing the nice!nanos.

bmcustodio, 2022-03-07:
Thank you very much @mattdibi, this is really helpful! If I decide to try my luck at building one, I will add the guid if it hasn't been added by then!

ocfox, 2023-05-17:
@snizovtsev Can you briefly describe how it needs to be done? I just got 3 QFAC 14015, and I want to continue to finish porting the Zmk firmware.

snizovtsev, 2023-05-17:
> @snizovtsev Can you briefly describe how it needs to be done? I just got 3 QFAC 14015, and I want to continue to finish porting the Zmk firmware.

It's not so easy. I had a barely working prototype a year ago and then give up because unplanned immigration eaten all of my free time. But I have much more time now and would like try dig into it again if its not just me interested in it.

The first thing to consider is that QFAC 14015 is not much better than QFAB. Before my QFAC chips arrived, I spent a lot of time on shrinking Zephyr's simple "bluetooth hello world" project to fit 16KB of QFAB. It almost worked but was so hacky and had no room for further applications.

With 32kb of QFAC things becomes much easier but it still a lot less than ZMK and Zephyr are designed for. A lot of time should be spent on optimizing the core of Zephyr's Bluetooth stack and ZMK for a 32kb memory chip.

I used `zephyr-v2.7.1` as base. I remember that other versions had some NRF51-related regressions and didn't initialize bluetooth. Then I shrinked all large parts that prevents compilation of ZMK for QFAC. I remember I've even needed to reduce number of keys in the keymap to fit into memory. Its all is so hacky that I've haven't made any commits and left it in form of dirty project on my computer. I will try find some time to recap and publish that patches soon if you need it.

ocfox, 2023-05-17:
> I remember I've even needed to reduce number of keys in the keymap to fit into memory.

It doesn't sound like there's much availability, and if that's the case then I guess there's no point in continuing this work.

harshitgoel96, 2025-03-12:
Hi, is there any further development on this?

I also wanted to know if using a ZMK dongle will help with clone YJ-14015's 16k limitations  somehow?

harshitgoel96, 2025-07-08:
I am not sure if someone else will be interested in this, but looks like we can use Seeed's nrf52840 Xiao for ZMK. 

Now by default Xiao has 11 GPIO pins listed. but looking at this [message](https://forum.seeedstudio.com/t/nfc-antenna-plugged-in-seeed-xiao-nrf52840/275004/15) it might be possible to repurpose the NFC pins to be used as GPIOs. 

If there is enough interest i can try working on a shim or something.