mattdibi/redox-keyboard #128: Falbatech PCB case compatibility

Decappi, 2022-01-16:
Hi,

The Wireless Rev2 case is not compatible with the latest Falbatech's PCB, as the moved diode positions do not nicely align with the bottom plate.
Right now I'm printing the Rev1 version.
Falba's updated PCB is based on the revision 1 with some of the hardpoints rotated, the bottom row moved up + Hot swap pads.
Would you be so kind to generate the STLs for this one as well? I've tried to do this myself, but I failed miserably in blender, so here I am, asking you for help :)

I've attached some pics of the plate, I could also scan it if you'd like a less skewed look:
[image]
[image]

mattdibi, 2022-01-17:
Hi @Decappi,

are you referring to [this](https://github.com/mattdibi/redox-keyboard/tree/master/redox-w/rev2.0WHS/case) case?

This case was designed using the KiCAD PCB files as a starting point. Not having access to Falbatech's ones (or one of their PCBs) this gets really hard to replicate.

Have you considered using Fusion360 instead of Blender? It should be easier for you to modify the .STEP with that.

Decappi, 2022-01-17:
Hi @mattdibi,

yes, I was referring to this case.
I'll try to contact Falbas for access to those files. Worst case I can send you one of the physical PCBs, as I have 6 of them right now.

About the fusion thing - I'm a software engineer, not a real engineer, so neither do I have Fusion360 licence, nor the real understanding of how to do this even if I have the software.

mattdibi, 2022-01-18:
> Worst case I can send you one of the physical PCBs, as I have 6 of them right now.

Yeah, currently I don't have the time nor the will to work on this. Don't worry about that.

> About the fusion thing - I'm a software engineer, not a real engineer, so neither do I have Fusion360 licence, nor the real understanding of how to do this even if I have the software.

I too am a software engineer and Fusion360 is free for hobbyist. Give it a try!

Decappi, 2022-02-18:
I've tried using the Fusion360. That's been a pretty soul crushing experience to say the least, as it doesn't exist for Linux, the wine version is broken and the browser version is just so slow I can't finish anything before the heat death of the universe.

As of now, there are no freely available printable cases that are compatible with the falbatech PCBs, so I've created a PR with the updates to the Readme, so that the next builders will know that they need to buy the cases from Falba in order to use the PCBs.

closed 2022-02-19 by mattdibi