mattdibi/redox-keyboard #130: 2.0WHS Case?

buddylindsey, 2022-02-20:
So i am trying to figure out what case to print for the V2 wireless.

I like the case that is https://github.com/mattdibi/redox-keyboard/blob/master/img/pcb-w-assembly1.jpg but not sure where the files for it are, or if it is compatible with the v2 WHS.

Is the case that is designed for the v2 just the open design like in this picture: https://raw.githubusercontent.com/mattdibi/redox-keyboard/master/img/redox-w-2.0-case-1.jpg

If so I am even still a bit confused on what I need to print and how to assemble that.

Sorry for the questions on this. I might just be missing something somewhere or in the docs. Just not sure what to print. Hoping I can get to actually soldering up everything this week and maybe put this together this week, I hope.

mattdibi, 2022-02-20:
Hi there,

> I like the case that is https://github.com/mattdibi/redox-keyboard/blob/master/img/pcb-w-assembly1.jpg but not sure where the files for it are, or if it is compatible with the v2 WHS.

That was a case designed and manufactured by Falbatech. I do not own the design files nor the means to reproduce it.

> Is the case that is designed for the v2 just the open design like in this picture: https://raw.githubusercontent.com/mattdibi/redox-keyboard/master/img/redox-w-2.0-case-1.jpg

That is the case designed for the Redox 2.0WHS and is available here: https://github.com/mattdibi/redox-keyboard/tree/master/redox-w/rev2.0WHS/case

There's also a [closed back design ](https://github.com/mattdibi/redox-keyboard/blob/master/redox-w/rev2.0WHS/case/BackplateL.stl)that I did not test.

> If so I am even still a bit confused on what I need to print and how to assemble that.

You need to print the [front plate](https://github.com/mattdibi/redox-keyboard/blob/master/redox-w/rev2.0WHS/case/FrontplateL.stl) and one of the back plates. Those are designed for the left hand, you'll just need to mirror them along the X axis on your slicer to get the right hand.

Assembly just need a few m3 screws and bolts.

Hope this helps...

buddylindsey, 2022-02-20:
Makes sense. I plan to make a case out of walnut someday so not really going to try too hard to re-create a bigger case.

I re-looked at the front and backplates I printed and figured out where my confusion was coming in. I was seeing a conflict where the battery holder etc. was going to conflict with the top part of the case. However, I realized I was putting the board in "upside down". Once I realized all the stuff will be under the pcb board in the case it clicked in my head and it works.

I printed the closed backplate and will try that first and let you know.

Final question is there a "suggested" keys for all the locations?

This is, literally, my first custom keyboard, and I don't know what keys might even go where. Up until now I have just bought logitech ergonmic keyboards. I especially am not sure for space bar and a key to change modes, I think that is a thing. Could you maybe put up your layouts as a starting point somewhere?

I told my wife I figure if I am going to get into a custom keyboard lets go hard core and print some pcb boards. lol

mattdibi, 2022-02-20:
> Final question is there a "suggested" keys for all the locations?

You can find [a starting keymap](https://github.com/qmk/qmk_firmware/tree/master/keyboards/redox_w/keymaps/default) in the [QMK firmware](https://docs.qmk.fm/#/) repository. I suggest you to take also a look at [the others](https://github.com/qmk/qmk_firmware/tree/master/keyboards/redox_w/keymaps) for ideas (or even [the Redox wired ones](https://github.com/qmk/qmk_firmware/tree/master/keyboards/redox/keymaps)).

Here's an example: https://github.com/qmk/qmk_firmware/tree/master/keyboards/redox/keymaps/jeherve.

Be sure to take a look at the [firmware section](https://github.com/mattdibi/redox-keyboard/tree/master/redox-w#firmware) of the Redox Wireless to understand how many different software parts are at play.

> I especially am not sure for space bar and a key to change modes, I think that is a thing.

I think you're talking about "[layers](https://docs.qmk.fm/#/feature_layers)".
For those I suggest you to take advantage of the thumb clusters. The keymaps I linked you should give you a few ideas.

buddylindsey, 2022-02-22:
Closing this as I got it built, now to figure out why things aren't working properly.

closed 2022-02-22 by buddylindsey