mattdibi/redox-keyboard #132: Battery Drain if soldered poorly

buddylindsey, 2022-03-09:
I just wanted to note an issue I ran into with the Rev2WHS. Might be worth adding a note to the docs somewhere, or close this and leave it here for if someone searches closed issues like I do.

I think I soldered the battery holder a bit too far back and the Hot Swap piece to far forward so that when I push the batter all the way in on one of my sides it touches the hot swap connector. I think this drained my batter in a matter of a few days because as soon as I swapped out the battery the keyboard worked just fine again.

Just wanted to bring it up as a problem that can happen if you aren't careful.

mattdibi, 2022-03-09:
Thank you for pointing this out. I would keep this issue here for documentation purposes as you suggested.

closed 2022-03-09 by mattdibi

Zielak, 2022-06-04:
Same happened here, but that's how my PCB was printed. One fix is to just place some insulating tape over that hot-swap socket.