mattdibi/redox-keyboard #136: Pro micro question

jmnguye, 2022-03-26:
I've ordered 2 PCB , and i dont understand why i have to flip the pro micro in order to get it work when connect

I've followed the installation and should have something like this https://raw.githubusercontent.com/mattdibi/redox-keyboard/master/img/rgb-underglow2.jpg

But when i do exactly the same, my pro micro is not working, however, if i flip my pro micro, there is a red light, and i'm able to reset.

https://imgur.com/a/3Mio35U

something is wrong somewhere, but i cannot find out where

phongnh, 2022-03-26:
@jmnguye Maybe the issue comes from USB-C Pro Micro, B1 and B3 are swapped. Please check this link https://kbd.news/USB-C-Pro-Micro-issues-1054.html

mattdibi, 2022-03-26:
@jmnguye In the pics the Pro Micro is not soldered to the board, it this how you tested it? Without (properly) soldering there's no guarantee that all the pins are actually contacting with the pads on the Arduino therefore I wouldn't trust the results of your test.

As to why you need to flip them, without going much into detail, this is a common way to use the same PCB for both left and right hand in split keyboards. The PCB is the same for both halves, since you need to flip one to have a left and a right hand you need to flip the Pro Micro accordingly. If you want more details take a look at the schematic.

Other notable examples of this technique: [Let's Split](https://github.com/nicinabox/lets-split-guide/blob/master/assembly.md), [Mircodox](https://boardsource.xyz/store/5f2e7e4a2902de7151494f92), [Architheutis Dux](https://github.com/tapioki/cephalopoda/tree/main/Architeuthis%20dux).

jmnguye, 2022-03-26:
Thx guys, the issue came from the RGB strip soldering. I've followed @mattdibi , and soldered correctly the Pro Micro, but noticed that from time to time the pro micro light was on.
I'll have to reorder and start over, thx for your help @phongnh  and @mattdibi

closed 2022-03-27 by mattdibi