mattdibi/redox-keyboard #137: Right half of Redox Handwire not recognized by VIA

wryman, 2022-04-12:
Hi!

I just finished wiring my redox (first handbuilt) and flashed both Pro Micro's with the default VIA firmware and cannot get the right half to be recognized when both halves are connected with the TRRS cable, both mc's power up when connected via TRRS just no keypresses are recognized on the right half.   Each half is recognized independently as a left hand module with all keys working when connected with USB.

I have wired the TRRS in the i2c configuration shown in the schematic with the 2-4.7k ohm pull ups located in the left hand module, I have tested continuity and resistance between all pins across the TRRS cable and everything checks out as expected. I have confirmed that my Pro Micro's both have the J1 jumper unbridged.  I have tried clearing EEPROM's, reflashing with the VIA hex and setting left/right handedness in EEPROM with QMK toolbox with no success.  I have also tried wiring the TRRS jacks in serial with the soft serial pin on the default pin2/pd0 to rule out issues with the pull-ups, but was met with the same result (right half not recognized).  

Am I missing a step somewhere? The only thing I can think of is that the VIA fw is missing some crucial function, I see that pin 2/PD0 is the soft serial pin set in the config files and I have confirmed that my pinouts are correct so I'm at a bit of a loss.  I don't have much experience with compiling custom QMK firmwares so I figure I'm missing a setting somewhere.  

FWIW, I do not have any RGB LEDs connected.

CameronWeller, 2022-04-12:
I have had the exact same issue ! I actually did a handwire build but the wiring is very similar, just with serial instead of I2C. What i ended up having to do was change the configuration to #define enable_split_usb_detect or something along that line. It seems the via build doesn't have the configuration right and its hard to change that because via is actually closed source. I am currently working on a vial (open source competitor) build. I can send you the relevant build if need be.

wryman, 2022-04-12:
Hey, Thanks! That'd be awesome if you could send the build.  At least that
way I can get it up and running and see what the differences between the
config.h files are and maybe try and work backwards from there.

CameronWeller, 2022-04-12:
totally, might be a while as there have been some hiccups in the setup as I am kind of a noob at the software side of things.

wryman, 2022-04-13:
> 

No worries, thanks for replying.

wryman, 2022-04-13:
> 

Bit of an update, I reflashed both halves with the default QMK keymap with the #define EE_HANDS parameter added to the config.h and am able to get each individual half working as the corresponding left/right when connected via USB separately.  However when I connect the TRRS cable and then plug the left half into USB, I now get no inputs whatsoever from either half.

wryman, 2022-04-13:
Well I figured it out! the Pro Micro clones I had purchased had the pads on J1 joined with a trace beneath the mask! I have some images that show the pads with and without the trace cut.  Cutting the trace between the J1 pads on the slave side fixed it for me! 

[https://imgur.com/ayUx8Q3](url)
[https://imgur.com/OKROYgI](url)

closed 2022-04-13 by wryman