mattdibi/redox-keyboard #140: Keyboard not working at all

AsatothAbaddon, 2022-04-21:
Hello everyone,

I have soldered this keyboard and flashed as described in the repository with no error, but the keyboard is not working at all.
I see the redox-wireless receiver in my settings in Windows, but no input is sent to the System.

Since i dont have the on/off switches right now, I have soldered the PCBs to be on, I hope it works this way as well?
Also the LEDs are not working, but I am pretty sure this cant be the issue that nothing works.
Yes, I am not the best in soldering work, but the previous keyboard, an Ergodox works.

I have uploaded some images here: https://github.com/AsatothAbaddon/redox-w

I hope anyone can help, I really want to finish this keyboard.

Thanks in advance!

mattdibi, 2022-04-21:
Hi,

> Since i dont have the on/off switches right now, I have soldered the PCBs to be on, I hope it works this way as well?

It should as long as the pads you joined are the correct ones. Can't quite make out if that's the case but a multimeter should tell you all you need to know. If you see 3.3V on the VCC pad of the YJ-14015 you're good. Another way to power the YJ-14-015 is through the programming headers using the ST-Link programmer.

> Also the LEDs are not working, but I am pretty sure this cant be the issue that nothing works.

Yeah, should not be a problem.

> Yes, I am not the best in soldering work, but the previous keyboard, an Ergodox works.

Sorry to tell you but those two are quite different in the skill required. The YJ-14015 is hard to get right and, judging from the pics, it might be a soldering issue from what I see.

Try powering the board through the ST-Link programmer (removing the batteries of course) and try it out. If it doesn't work it's most probably a soldering issue.

AsatothAbaddon, 2022-04-22:
Thanks for your reply!

So I think i need a multimeter.

> Try powering the board through the ST-Link programmer (removing the batteries of course) and try it out. If it doesn't work it's most probably a soldering issue.

I tried to use the 3.3v and the GND Pins with the corresponding onces on the PCBs (ofc, with no battery) and while the receiver is plugged in but nothing happend...

I order a multimeter right now and make tests on the weekend.