mattdibi/redox-keyboard #143: Right side don't work

brunooomelo, 2022-05-12:
hi, i followed the tutorial, the left side of the keyboard is correct but the right side is not detecting the keys

mattdibi, 2022-05-13:
Is it a wired keyboard? If so, did you check connectivity on the TRRS cable? Did you check soldering issues on the TRRS connector? On the Pro Micro? Did you try the pro micros before installing (especially the one that's not working)? Did you burn the firmware on both pro micros?

brunooomelo, 2022-05-14:
I did all the procedures, I installed it correctly, I checked the connectivity, I burned the firmware on both pro micro, only the right side doesn't work, it lights up the led but the keys don't work

mattdibi, 2022-05-14:
Are you using I2C or serial for connecting the two halves? On the latest builds of QMK there seems to be an issue with I2C.

In any case the most common type of issue is with the hardware. Please check for cold joints and/or soldering issues on the board especially around the TRRS connectors.

brunooomelo, 2022-05-14:
I tested the connection of the TRRS between the sides and it is normal

brunooomelo, 2022-05-16:
can I use serial instead of i2c without rgb?

mattdibi, 2022-05-16:
RGB only works with serial since one of the wires of the trrs cable is used to carry the data signal for RGB. This is explained [nere](https://github.com/mattdibi/redox-keyboard/tree/master/redox#rgb-underglow-firmware).

closed 2022-07-27 by brunooomelo