mattdibi/redox-keyboard #146: [Help] redox w can’t send any key presses

jjwspring, 2022-05-31:


jjwspring, 2022-06-05:
To refine the question, Is there an easy way to check whether the nrf51822’s are working properly. And if not to find out why they are not working?

mattdibi, 2022-06-10:
Hi @jjwspring,

> Im currently trying to build a redox w keyboard, but wanted to hand wire it, (so no pcbs).

I see you like challenges, I respect that.

> I’m using core51822’s but no the 51822(B) would that make a difference?

I developed the first prototype of the Redox W with the full size Core51822. There's no difference HW-wise It's just a different form factor. In this case the Core51822 is a good choice since it's more suitable for a hand wired board.

To further proceed in the debug process I would recommend you to double check [how the firmware works](https://github.com/mattdibi/redox-keyboard/tree/master/redox-w#firmware), [the schematics](https://github.com/mattdibi/redox-keyboard/tree/master/redox-w/rev1.0W/pcb) and then further dig into the various steps in the chain: does the arduino receives data from the Core51822 running the receiver code? If so, does the Core51822 running the receiver code receives data from the Core51822 running the transmitter code? Try modify the firmware and log whatever you can for further debugging.

jjwspring, 2022-06-14:
Hi @mattdibi 
Thank you for your reply, and for making this project!
It’s good to hear that the only difference between the core51822s is the form factor, that’s what I was hoping!
How would I go about editing the firmware on the Core51822s? If I edit the files in my clone of the redox-w-firmware folder and then follow the flashing procedure will the changes be included, or do I need to do an extra compiling step?