mattdibi/redox-keyboard #147: [Help] Errors compiling firmware

AlexanderHOtt, 2022-06-06:
I'm trying to compile the firmware on windows but I keep getting errors.

I followed the instructions on how to setup QMK but when I try to compile the firmware I get the error
```
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `''
/usr/bin/sh: -c: line 2: syntax error: unexpected end of file
make[1]: *** [.build/obj_redox_rev1_default/obj.txt] Error 2
Make finished with errors
make: *** [redox/rev1:default:avrdude] Error 1
```

<details>
<summary>
Full output of <code>make redox/rev1:default:avrdude</code>
</summary>
$ make redox/rev1:default:avrdude
QMK Firmware 0.17.0
Making redox/rev1 with keymap default and target avrdude

avr-gcc.exe (GCC) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_redox_rev1/src/info_config.h                                                 [OK]
Generating: .build/obj_redox_rev1/src/default_keyboard.c                                            [OK]
Generating: .build/obj_redox_rev1/src/default_keyboard.h                                            [OK]
Generating: .build/obj_redox_rev1/src/layouts.h                                                     [OK]
Compiling: keyboards/redox/redox.c                                                                  [OK]
Compiling: keyboards/redox/rev1/rev1.c                                                              [OK]
Compiling: .build/obj_redox_rev1/src/default_keyboard.c                                             [OK]
Compiling: keyboards/redox/keymaps/default/keymap.c                                                 [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/send_string.c                                                                    [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/sync_timer.c                                                                     [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/bootmagic/magic.c                                                                [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_defer_g.c                                                           [OK]
Compiling: quantum/split_common/split_util.c                                                        [OK]
Compiling: quantum/split_common/transport.c                                                         [OK]
Compiling: quantum/split_common/transactions.c                                                      [OK]
Compiling: quantum/main.c                                                                           [OK]
Compiling: quantum/mousekey.c                                                                       [OK]
Compiling: quantum/color.c                                                                          [OK]
Compiling: quantum/rgblight/rgblight.c                                                              [OK]
Compiling: quantum/process_keycode/process_rgb.c                                                    [OK]
Compiling: platforms/avr/drivers/ws2812.c                                                           [OK]
Compiling: quantum/led_tables.c                                                                     [OK]
Compiling: quantum/crc.c                                                                            [OK]
Compiling: quantum/process_keycode/process_magic.c                                                  [OK]
Compiling: quantum/command.c                                                                        [OK]
Compiling: quantum/process_keycode/process_grave_esc.c                                              [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Compiling: platforms/avr/drivers/i2c_master.c                                                       [OK]
Compiling: platforms/avr/drivers/i2c_slave.c                                                        [OK]
Compiling: platforms/avr/drivers/serial.c                                                           [OK]
Assembling: platforms/avr/xprintf.S                                                                 [OK]
[... 26 lines cut ...]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 [OK]
Compiling: tmk_core/protocol/lufa/usb_util.c                                                        [OK]
/usr/bin/sh: -c: line 1: unexpected EOF while looking for matching `''
/usr/bin/sh: -c: line 2: syntax error: unexpected end of file
make[1]: *** [.build/obj_redox_rev1_default/obj.txt] Error 2
Make finished with errors
make: *** [redox/rev1:default:avrdude] Error 1

</details>

I've also tried using `qmk` to compile it.
```
qmk config user.keyboard=redox/rev1
qmk config user.keymap=default
qmk compile
```
but I get the error
```
WARNING "git describe --abbrev=6 --dirty --always --tags" returned error code 127
WARNING "git describe --abbrev=6 --dirty --always --tags" returned error code 127
WARNING "git describe --abbrev=6 --dirty --always --tags" returned error code 127
builddefs/build_keyboard.mk:49: *** multiple target patterns.  Stop.
Make finished with errors
make: *** [redox/rev1:default] Error 
```

mattdibi, 2022-06-10:
Hi @AlexanderHOtt,
this looks like a QMK problem more than a Redox problem. Take a look [here](https://www.reddit.com/r/olkb/comments/n7bj77/qmk_not_compiling_property/) and [here](https://github.com/qmk/qmk_firmware/issues/6094) and see if it helps you. If this doesn't help you I recommend you to ask to the QMK maintainers.

AlexanderHOtt, 2022-06-16:
@mattdibi thank you for the suggestions. I ended up doing a fresh install of QMK MSYS and using the QMK MSYS terminal instead of the MSYS2 MINGW terminal I was using before and that fixed all my problems.

closed 2022-06-16 by AlexanderHOtt