mattdibi/redox-keyboard #149: [docs] Info that I've discovered about the elite c microcontroller

JamesYeoman, 2022-06-13:
So I finally have a working redox keyboard after a lot of incompetency on my part. I noticed you don't have any info about the Elite C, and my keyboard is using the Elite C, so I thought it would be good to share my findings.

First of all, despite what Pastitas told me on the issues page on their repo, the [neodox case](https://github.com/Pastitas/Redox-neodox-Case) does have just enough room for the Elite C. The Pro Micro certainly wouldn't have fit in there, but the Elite C just about fits. And that's including the fact that I'm using the standoffs that came with it.

Secondly, when flashing the Elite C, avrdude doesn't detect the keyboard, but dfu-programmer detected it perfectly. I haven't tested out EE_HANDS yet, but MASTER_LEFT is working like a treat.

If there's anything else you want to know, just let me know. But I thought these would be particularly useful for you, and sufficient enough for anyone who wants to use an Elite C.

JamesYeoman, 2022-06-15:
I've since found this https://github.com/qmk/qmk_firmware/issues/16450 which mentions about how the Elite C uses a different bootloader altogether, but it doesn't explain why when the firmware was built for catarina (`make redox/rev1:<<keymap>>:avrdude`), it still works with dfu-programmer... very odd, but now my flashing method is simpler. So basically, in order to support Elite C, people need to make their own fork of qmk, and create their own keymap, in order to create a `rules.mk` and add `BOOTLOADER = atmel-dfu` to it.

DafDandy, 2022-08-17:
I am also using the elite c on both halves of my keyboard but was unable to flash it with the qmk cmd line. I'm extremely confused because the boards work on their own but when connecting them together with the trrs cable they seem to lose power and turn off.

Is it possible that the elite c doesnt need the resistors and that it causing too much power drop? Sounds like you've gotten yours working and might be able to offer some invaluable advice to a struggling diy experience

DafDandy, 2022-08-17:
i've confirmed on the side with the resistors that at the trrs jack its outputting 5v but as soon as i plug in the trrs cable even without the other board it drops to nothing. This is both with and without hot-swap testing

mattdibi, 2022-08-17:
@DafDandy "USB device requires more power than can be delivered" I think this is an evident symptom of a short somewhere. Ensure your TRRS cables are plugged all the way in and look for shorts in the TRRS connector area.

Furthermore what you describe is quite different from the original intent of this issue. Please open a new one if you still have issues.