mattdibi/redox-keyboard #153: diodes were soldered wrong direction

UrielParienti, 2022-08-24:
Hi,
i finished the left side (work perfectly)
but I am facing issues with the right side.
I'm assuming that the diodes were soldered wrongly, but before I cut them all out I want to be sure that this is the issue.
all of them are soldered with a little black mark on the right side.
when I started uploading the firmware, it finished successfully but non of the keys were working.
so am I doomed to remove them all?

mattdibi, 2022-08-24:
Hi there,
if you could upload a picture it would help.

Diodes need to be installed following the symbols that are printed on the PCBs like in the image below:
[image]

Anyway you can also refer to the images in this repo for reference.

[image]
[image]

UrielParienti, 2022-08-24:
so its definitely the issue 

[image]

ill start cutting them off.
hope I did not damage the controller when I desoldered it.

Thanks a lot

closed 2022-08-24 by mattdibi