mattdibi/redox-keyboard #156: Question: Can you use LIR2032 in the wireless version?

neopostmodern, 2022-10-04:
Has anybody tried using LIR2032 instead of CR2032? I'm having connection issues as described in #55 and a higher voltage mitigates most problems – so rather than frequently replacing CR2032 and creating a bunch of waste I considered having a rechargeable solution which I could just recharge as soon as I experience connectivity troubles.

My concerns are:
- Is the maximum output of 3.7V of a LIR2032 too much? (I think a fresh CR2032 maxes at 3.2V?)
- Can I expect a reasonable time between charges given the much lower capacity of a LIR2032?

Hints and shared experiences much appreciated!

(My alternative thought is building a [CR2032 emulator](https://www.tindie.com/products/bobricius/coin-cell-battery-emulator-cr2016cr2032/) but then I end up having a wired keyboard afterall...)

marco73, 2022-10-29:
@StoyanDimitrov, My boards are also consuming/burning through CR2032 at a weekly rate. At the end of the week some keys are beginning to response badly, so I was thinking about the using another source of power, like a lipo battery. But the solution you mention is still using 4,2 V when the lipo is fully charged. According to the datasheet of the YJ-14015 states it can handle 2 - 3,6V.  How does the YJ-14015 handle the 4,2 V, or what did you do to prevent a higher voltage than 3,6V?

StoyanDimitrov, 2022-10-31:
The batteries are rated to 3.7 V. 4.2 V is the charging voltage meaning this is the amount that you charge the batteries with and not the one that the battery will can give out when fully charged.
> According to the datasheet of the YJ-14015 states it can handle 2 - 3,6V. How does the YJ-14015 handle the 4,2 V, or what did you do to prevent a higher voltage than 3,6V?
Nothing, just substituted the CR2032 and its holder with charging module and li-po battery. I guess the YJ-14015 can "just handle" the higher (0.1 V) voltage.

In #55 there were other ideas about substituting the original powering element with an alternative one but choices are dependant of the keyboard case you have.

marco73, 2022-11-08:
I found these [little boards](https://www.aliexpress.com/item/1005004680099594.html?spm=a2g0o.cart.0.0.71c038daTpJErS&mp=1). It takes a lipo and outputs a stable 3.3v. Hope this will solve the intermittent response from some keys. I noticed the same issue as described on the latest revision of the [Aerodox](https://hackaday.io/project/171535-aerodox) and he resolved it with an similar solution.

neopostmodern, 2023-03-18:
@marco73 Can you report how you've been faring with the mentioned boosters? 
What capacity of LiPo are both of you using and how long can you go between the recharges?

neopostmodern, 2023-03-19:
Also, @StoyanDimitrov – what is the danger of LiPos? Aren't they in every other gadget?

StoyanDimitrov, 2023-03-23:
1800 mAh per half. It looks like this (it's an unprotected pack, never use unprotected packs!):
[image]
It's 2.5 mm thick. The thickest one that fits under the PCB in my case. Actually the legs of the keys are longer than the PCB and if they stick out they could damage the battery pack. Look for that if you think about something similar.

Fully charged the batteries work for months. Like 3-4-5 months (don't remember, don't blame me) at full 4+ hours typing a day. Charging time is about an hour or so. The charging modules have blue and red LEDs. Red for charging, blue when ready.

Keep in mind that the "original" YJ-14015 - the [Core51822B](https://www.waveshare.com/wiki/Core51822_(B)) has different dimensions and may not fit the pads of the PCB.

As for the dangers of the li-pos you may want to educate yourself.

snizovtsev, 2023-10-03:
I'm using a batch of cheap Chinese LIR2032 batteries daily for many years and they work just fine.

PhiBabin, 2023-10-28:
I made [a PCB that replaces the coin cell battery](https://github.com/PhiBabin/Redox-Lipo-Adapter), it has a USB-C connector and circuitry to charge a LiPo battery.