mattdibi/redox-keyboard #157: LED of redox-w keyboard receiver does not work

Sp-Link, 2022-11-23:
Hi @mattdibi 
I made the  redox-w keyboard according to the document, and tested that the keyboard works normally, but the LED indicator of the receiver does not blink.I searched the project file, but I couldn't find the description of LED functions.What is the function of the LED on the receiver? When does the LED  blink? I hope I can find the answer from you.

mattdibi, 2022-11-23:
Hi @Sp-Link,

the LEDs status is set [here](https://github.com/qmk/qmk_firmware/blob/558ab18611665c3b447c27bac82fac2224eadaef/keyboards/redox_w/keymaps/default/keymap.c#L91). They are activated and set depending on the layer you're currently on. By deafult they're set to off.

More info [here](https://github.com/mattdibi/redox-keyboard/issues/93#issuecomment-808697030).

Sp-Link, 2022-11-24:
> Hi @Sp-Link,
>
> the LEDs status is set [here](https://github.com/qmk/qmk_firmware/blob/558ab18611665c3b447c27bac82fac2224eadaef/keyboards/redox_w/keymaps/default/keymap.c#L91). They are activated and [...]

> 

Thank you very much for your help. The problem has been solved.This design is very ingenious!

closed 2022-11-24 by Sp-Link