mattdibi/redox-keyboard #158: Help - One Sided Redox

evidenz, 2022-12-07:
Hello there, 

Is it possible to simply use just one side of the redox? If so, are there any caveats to it that I should keep in mind? To be precise, I have already built one but it seems that keys are not registered properly and I assume it is because the other side is missing and hence the whole layout matrix cannot be translated properly?

mattdibi, 2022-12-07:
Hi @evidenz,

yes it is!

If I had to guess your Redox is probably not detecting which is the correct master hand and thus mirroring the layout. Take a look at the QMK documentation [here](https://docs.qmk.fm/#/feature_split_keyboard?id=setting-handedness).

evidenz, 2022-12-07:
Ah, I will have a look. Currently VIA is flashed :) do you know by chance if this is also required/possible there to set the master? Otherwise, I have no problem going with QMK :)

mattdibi, 2022-12-08:
Unfortunately no, I don't know of a way to set handedness with VIA.