mattdibi/redox-keyboard #159: Help - About PCB and Redox Manuform

Jetiaime, 2022-12-30:
Hello there,

I'm a hardware rookie, and I've never touched anything about hardware before. But I think the Redox Manuform model is really nice, and I would really want to have a keyboard like this of my own. So I would like to ask: Can the PCB from this project be used directly for the Redox Manuform model or what modifications do I need to make to make the provided PCB suitable for the Redox Manuform?

Chrismelba-Instaclustr, 2022-12-30:
> NetApp Security WARNING: This is an external email. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Looking at that I think it will be a fully hand wired keyboard. You don't use a pcb, just lots of wires and patience. A tricky project for your first hardware, but not impossible!

mattdibi, 2022-12-30:
Hello @Jetiaime,
as @Chrismelba-Instaclustr already said the Redox Manuform is a handwire project. The Redox PCBs won't fit in the case and can't be used directly.