mattdibi/redox-keyboard #165: Optical Switches?

ninjrdevelop, 2023-03-25:
Would the Rev 2.0WHS work with Optical switches? E.g: https://www.amazon.com/Gateron-Ks-15-Optical-Mechanical-Switch/dp/B083F44F5R

The listing states

> COMPATIBILITY – Optical Switches are compatible with Optical Keyboards and can be hotswapped without desoldering. They are available in 65 PCS, 90 PCS and 120 PCS making them the ideal fit for 60%, TKL and full size keyboards. Gateron Optical switches are not compatible with regular mechanical keyboards.

But I'm not too sure the difference between optical and normal switches.

mattdibi, 2023-03-25:
@ninjrdevelop no they won't work.

These switches work on with fundamentally different principle than traditional switches. It requires specific support on the PCB level for this to be used and this is not currently supported for the Redox keyboard.

ninjrdevelop, 2023-03-26:
Roger, thanks for the clarification.

closed 2023-03-26 by ninjrdevelop