mattdibi/redox-keyboard #168: I2C communication is not working

paradogk, 2023-05-06:
Hello, there.

Thanks for matt, I really hope overcoming my stiff shoulders.

However, I faced some issue on my keyboard. I set EE_HANDS (config.h) as describing github.

first, I flashed eeprom to each side. left side flashed left-eeprom and right side flashed right-eeprom bin

and then, I connected both side with TRRS cable(4 pole) and pluged USB cable from left side to PC.

PC recognized keyboard and left side button was working nomally,  but PC could recognized right side button.

I already tested each side button working by connecting USB on each side. 

I Think I2C communication is not working . therefore Is anybody who can help me to overcome my stiff shoulders.

Thanks

paradogk, 2023-05-08:
To help someone and leave the history.

I found someone issue like me on this github issue.

PRDeltoid, 2023-10-02:
Took me a while of searching before finding this `#define SPLIT_USB_DETECT` solution (via the QMK discord). I am curious what downsides or tradeoffs there are for using SPLIT_USB_DETECT versus the default detection method. And what caused this #define to be start being needed? Was it a QMK change?