mattdibi/redox-keyboard #170: Question: How does redox wireless handle

jadewolf05, 2023-06-15:


StoyanDimitrov, 2023-06-15:
This is done on firmware level. So every keyboard in an area have to have different [channels](/mattdibi/redox-w-firmware/blob/master/redox-w-keyboard-basic/main.c#LL32C25-L32C25) settings in the firmware or something like that.

jadewolf05, 2023-06-15:
> This is done on firmware level. So every keyboard in the area have to have different [channels settings](/mattdibi/redox-w-firmware/blob/master/redox-w-keyboard-basic/main.c#LL32C25-L32C25) in the firmware or something like that.

Thanks! I'll take a look at this once I get my second set built.

jadewolf05, 2023-06-16:
One more question in the same realm.   I went ahead and built the boards and flashed thew firmware but I noticed after flashing the arduino, both recievers ended up with the same id:
HID console connected: Mattia Dal Ben Redox Wireless (**4D44:5257:0001**)
Its unlikely both receivers will be used on the same PC so likely not a problem, but wanted to check if this is expected or if I need to do anything.

jadewolf05, 2023-06-16:
just digging in a bit and it looks like this info is written into the info.json file.

closed 2023-06-16 by jadewolf05