mattdibi/redox-keyboard #175: To migrate from ProMicro to RP2040-Zero.

azhizhinov, 2023-09-18:
Hello.
Can some one review the schema?
May be need some elements change, or other improvements?
Firmware should be the same, but recompiled with promicro_rp2040 converter.
Thanks in advance.
[20230918.REDOX-RECEIVER.00.pdf](https://github.com/mattdibi/redox-keyboard/files/12650147/20230918.REDOX-RECEIVER.00.pdf)
[image]
[image]

azhizhinov, 2023-12-08:
Hello.
No, I haven't ordered it, so there were no tests, nor firmware.
[20230925.REDOX-RECEIVER-ZERO.00.zip](https://github.com/mattdibi/redox-keyboard/files/13620534/20230925.REDOX-RECEIVER-ZERO.00.zip)

weekinro, 2024-02-25:
Hello, may I ask if the verification of RP2040 has been successful?

azhizhinov, 2024-02-26:
Hello.
Not really. Haven't ordered this PCB.
Sorry.

weekinro, 2024-02-27:
> 你好。 并不真地。还没有订购此 PCB。 对不起。

I have a PCB in my hand, but I don't know how to write code for qmk. Can you help me write code for me to verify?

azhizhinov, 2024-02-27:
Hello.
I've tried to modify the redox_w code for RP2040.
I'm failed with compilation.
I've found this [bug](https://github.com/qmk/qmk_firmware/issues/17744) and one man who can help us. :-)
Hey, @karosc, can you help us? Can you review the PCB in this issue?
Is "UART_DRIVER = vendor" in the rules.mk can fix the compilation process, or I need more changes?
Thanks in advance.
[20240227.REDOX_W.00.zip](https://github.com/mattdibi/redox-keyboard/files/14416131/20240227.REDOX_W.00.zip)

azhizhinov, 2024-05-10:
Hello there.
As I can see one of us has successfully compiled the firmware.
https://discord.com/channels/440868230475677696/1238328929924943883

weekinro, 2024-05-10:
> 你好呀。 正如我所看到的，我们中的一个人已经成功编译了固件。 https://discord.com/channels/440868230475677696/1238328929924943883

thanks,bro

azhizhinov, 2024-05-10:
Well, I don't know how, but I've managed to compile to.
Is someone is ready to test? Just remove ".zip" at the end of the file.
[redox_wireless_rp2040_default.uf2.zip](https://github.com/mattdibi/redox-keyboard/files/15275716/redox_wireless_rp2040_default.uf2.zip)
Sources is here.
[20240510.REDOX.WIRELESS_RP2040.00.zip](https://github.com/mattdibi/redox-keyboard/files/15275721/20240510.REDOX.WIRELESS_RP2040.00.zip)

azhizhinov, 2024-05-10:
If it's will work we can try to push that to the QMK.

azhizhinov, 2024-05-10:
There is similar project. Pierce.
I've made the receiver's PCB with dual compatibility. Both, RP2040-Zero and ProMicro can be used, not the same time, of course.
This receiver a bit different in pinouts, but I think there is the possibility to create an universal PCB, where the proper routing for Pierce or Redox can be switched with soldered jumpers.
https://github.com/durken1/pierce/issues/10