mattdibi/redox-keyboard #181: [rev2.0WHS] Receiver and Right Hand Transmitter Soldered

aliasboink, 2024-09-19:
Hello, 

I am rather new to soldering on PCB's and flashing firmware. I have done the following: 
- I have soldered the transmitter and the right hand transmitter. 
- I have flashed the QMK firmware on Arduino (2 Pro Micro 5V Arduinos, tested with both).
- I have flashed the firmware on the transmitter and the receiver using the docker compose setup.
- Once the transmitter is plugged in, only the Arduino green LED lights up.

Currently no key presses work on the keyboard. I have checked the connections with a multimeter and currently I am at a loss.
Can the firmware flash without the YJ-14015 module being soldered on properly?
How could I proceed with troubleshooting from this point onward?

aliasboink, 2024-09-19:
I've measured several things, among those being the voltage on the module. It's roughly 2.9V on the transmitter PCB and about 3.33V on the receiver (VCC to GND on YJ-14015)

aliasboink, 2024-10-02:
The columns all have roughly 3V. 
The rows all basically ground (pull down resistor I presume?).
Every column goes to 0 upon a keypress, specifically testing with K35 in my case for the most apart.
Any idea what could be the issue?

StoyanDimitrov, 2024-10-06:
Do you happen to have stable power source (i.e new batteries)? Lately the YJ-14015 modules are power-hungry.

aliasboink, 2024-10-06:
Yep, using new batteries, they measure 3V.